pschatzmann/codec-helix #1: Invalid framesize

podaen, 2021-12-10:
As I was saying in audio tools I have some troubles with playing some files. Convert the doesn't solve the problem soo that's why I make topic here.

> I did some tests with more than 100 mp3 files and here are the results...
> 
>1) The fast skipping forward songs are bad quality songs and could easily eliminated.
>2) The song where are cracks in it is a difficult one... The are all Lame encoded and above the 192kbs, but there are a few good songs above the 192kbs, so I could not point you in the right direction on bitrates. The only thing I could say is they are Lame encoded and can be of a good quality.

1) All those files have a low bitrate
2) I discoverd it is codec indipended. All those songs have an invalid frame size. 

It is clear that framesize can be diffrent. I did a quick search and I found this...

http://blog.coryhill.net/2009/06/calculating-mp3-frame-length.html

They say that in reality framesize can be diffrent. What framesizes do you handle?

podaen, 2021-12-10:
Here they go deeper in to the framesize

https://titanwolf.org/Network/Articles/Article?AID=25550046-6a8f-4386-90eb-a3098bb9ac8b

podaen, 2021-12-10:
I will post all frame size of 1 folder with trouble

podaen, 2021-12-10:
[image]
[image]
[image]
[image]

podaen, 2021-12-10:
The framesize changes in the files form 1044 to 1045 to 1044 to ...

pschatzmann, 2021-12-10:
https://github.com/pschatzmann/arduino-libhelix/blob/main/src/MP3DecoderHelix.h
#define MP3_MAX_FRAME_SIZE 1600 

The decoder tries to find the next SynchWord in the stream. It usually struggles if it finds some metadata instead of audio, but it then just continues with the next frame...

podaen, 2021-12-10:
> It usually struggles if it finds some metadata instead

Those only occure in the beginning or the end of the file. It changes all over the file.

I have some data that can find precise the end and the bigining of the next frame.

pschatzmann, 2021-12-10:
Not sure if it helps but you can try to increase
#define MP3_MAX_OUTPUT_SIZE 1024 * 5
#define MP3_MAX_FRAME_SIZE 1600

podaen, 2021-12-10:
On the first view I will say no, but I will try it and I let you know if ith helps.

podaen, 2021-12-10:
Where do I need to decare it?

podaen, 2021-12-10:
Ok. I found it at MP3DecoderHelix.h.

podaen, 2021-12-10:
Not sure in what changing it... The frame size is 1045 or 1044 and is less than 1600

podaen, 2021-12-11:
Soo what I get is that you are looking for sync words to determ the beginning and end of the frame. but I get the message of invallid frame size a lot. Would it not be beter to calclulated the frame size like this

Frame Size = ( (Samples Per Frame / 8 * Bitrate) / Sampling Rate) + Padding Size

pschatzmann, 2021-12-11:
I am not sure - but maybe the issue is that you get an error message - because this also has a negiative impact. So maybe it just helps to prevent messages in libhelix

podaen, 2021-12-11:
Than how could I prevent it?

pschatzmann, 2021-12-11:
I think you can add a 
#define HELIX_LOGGING_ACTIVE false
to your sketch

podaen, 2021-12-11:
I had to added it to the helix_log.h to make it work. Still have that issue in the middle of a file.

in those message is was saying the framesize was 0 or -1. Can't miss a frame to play to song correctly I think... And that is what I am hearing, fast skipping forward. Other ideas?

podaen, 2021-12-11:
hmmm, I tried this now and it's getting a lot beter now.

//#define MP3_MAX_OUTPUT_SIZE 1024 * 5
#define MP3_MAX_OUTPUT_SIZE 2048 * 5

//#define MP3_MAX_FRAME_SIZE 1600 
#define MP3_MAX_FRAME_SIZE 3200

closed 2021-12-11 by podaen

z-l-p, 2023-01-02:
Thanks for doing this research! I made an internet radio receiver based on the icystream example from the audio-tools library.  Worked great, except some stream URLs produced audio dropouts and invalid framesize warnings. I applied podean's changes to arduino-libhelix / src / MP3DecoderHelix.h and the problem was solved 100%. Maybe these could be the defaults in future releases? Or settable as decoder parameters in the main sketch?

pschatzmann, 2023-01-15:
I am a little bit reluctant to increase the default values because this increases the RAM use considerably.
You can call the following methods in your sketch to increase these values:
- setMaxFrameSize(int size)
- setMaxPWMSize(int size)

z-l-p, 2024-04-28:
Here is a working example of how to use the new setMaxPCMSize and setMaxFrameSize methods with a typical set of streams in arduino-audio-tools. I add it here because it took me awhile to figure out the syntax when libhelix is used within the EncodedAudioStream. I hope this is correct, and that it will help somebody. (It seems to work in my application).

**The constructors:**
_Streams: ICYstream "url" -> EncodedAudioStream "decMP3" -> VolumeStream "volume" -> I2SStream "i2s"_

```
ICYStream url; // called without (ssid, password) because I already set up
I2SStream i2s;
MP3DecoderHelix mp3;
VolumeStream volume(i2s); // for volume control option
EncodedAudioStream decMP3(&volume, &mp3); // Decoding stream
StreamCopy copierMP3(decMP3, url); // copy mp3 url to decoder (and push updates to dependent streams)
```

**The setup**
_I only show the decoder setup here. Other streams are standard as per examples._

```
decMP3.begin(); // start decoder
mp3.setMaxPCMSize(2048 * 5); // default is 1024 * 5
mp3.setMaxFrameSize(3200); // default is 1600
```