pschatzmann/codec-helix #5: low bitrate playback quality issues

goatzillax, 2023-11-16:
I've been playing around with the arduino-audio-tools to use it for audio notification with MP3 files through a bluetooth speaker.

My test project is here:

https://github.com/goatzillax/a2dp_mp3/

It's really just the player example, but I'm playing from LittleFS on an ESP32 (MH-ET Minikit).

The source files are pretty low bitrate -- 32kb.  By default they're practically unplayable, but I noticed one of the other issue reports regarding max frame size, so I tried the opposite and lowered the frame size and that improved things considerably.

However:

1. This seems pretty fragile
2. There are still audible artifacts like popping and slight stuttering

Is there anything else to try?  WAV files straight of RAM seem fine.

pschatzmann, 2023-11-16:
I never even tried to go that low: after all the lowest range that is offered by Audacity is 145-185!
Google gives the following: The MP3 format can range from around 96 to 320Kbps!

So either use a more normal bitrate or try the other codecs

closed 2023-11-16 by pschatzmann

goatzillax, 2023-11-16:
OK, I updated the bitrate to 96k.  It still seems to pop and stutter.

I'll poke around the other codecs.

pschatzmann, 2023-11-16:
If you are looking for an efficient codec on microcontrollers: adpcm might be a better choice...