pschatzmann/codec-helix #6: Misleading log output

stevstrong, 2024-02-23:
Hello,
I am using the `output_mp3.ino` example on F411 discovery board and I have set the LOG level to debug.
A portion of the output is this:
```
libhelix - synchronizeFrame
libhelix - -> frameRange -> 0 - 82
libhelix - -> we are at beginning of sync word
libhelix - decode 82
libhelix - -> bytesLeft 1600 -> 1183  = 417 
libhelix - -> End of frame (82) vs end of decoding (417)
libhelix - => provideResult: 2304
```
In particular, the line specifying the end of frame (82) and end of decoding (417).
Could you please explain why is there a difference?

I know that the previous frame was 417 byte long, but does this mean that there will be more bytes used for decoding this particular frame than the size of this frame (82 bytes)?

I have studied the code in function `decode` in` MP3DecoderHelix.h`, but I really don't understand the meaning of the variable `bytesLeft`
from which the value of decoded is calculated.
I mean why needs the decoder to handle how many bytes are left in the buffer? It should not be its task, but of the layer above.

pschatzmann, 2024-02-23:
If I remember right the decoder grabs as much as is able to process and reports back how may bytes it has processed. 
In my logic I just make sure that if I submit some data to decode, I have at least one valid frame.
From the reported processed end position, I know how many data I can remove from the buffer.

closed 2024-02-23 by pschatzmann

stevstrong, 2024-02-23:
Sorry to be picky, I just want to understand why the decoder could use for decoding more bytes than specified in the current frame.

I have read about bit reservoir, and I can see that the info about bit reservoir is only available for AAC (ADIF bufferFull), and is not defined for MP3.
I also see here some handling:
https://github.com/pschatzmann/arduino-libhelix/blob/27d09c34bd7edbaa1c53edf61fcd9bcbef5af523/src/libhelix-mp3/mp3dec.c#L339-L392
but there the `bytesLeft` calculation is mixed somehow with the available/used bytes from the `mp3DecInfo->mainBuf`, which has normally nothing to do with the read bytes from the _current_ frame of the stream.
So I really have doubt about the meaning of this variable.

pschatzmann, 2024-02-23:
bytesLeft is the total amount of available memory in the buffer that can be used for decoding.

stevstrong, 2024-02-23:
The link I posted points to` mp3dec.c` from `libhelix-mp3` subfolder, not `MP3DecoderHelix.h` ;)

pschatzmann, 2024-02-23:
Well, as I said - that code is not from me so I am not commenting on this.
I am just a user of libhelix and honestly don't bother about the implementation as long as it is working...