pschatzmann/codec-helix #11: Frame drops due to too little data in the buffer

VellingaT, 2024-03-27:
LS,

thanks for the MP3Decoder solution and sharing it.

Issue which I had and more or less "solved":
Feeding MR3Decoder with smaller amounts per call ( write 512 bytes per call) causes a lot of frame drops.
This can be prevented by first checking in case of valid header that there is enough data in the buffer.

I solved it by adding a check before starting decoding in "mp3dec.c"adding following two lines at position 305:
     int needed_bytes = mp3DecInfo->nSlots + 32 + 4 ; //  should come from mp3DecInfo sidebytes, header, CRC, padding
     if (needed_bytes > *bytesLeft)  return ERR_MP3_INDATA_UNDERFLOW;     // If not enough bytes in buffer exit without error

and in "MP3DecoderHelix.h"an additional test on line 141:
     if (result != ERR_MP3_INDATA_UNDERFLOW) {

Perhaps you can fix that in a correct way (special needed_bytes calculation) so others don't run into the same issue.

Regards, TV.

pschatzmann, 2024-03-27:
Thanks for your feed-back. Could you please double check with the [version in the development branch](https://www.pschatzmann.ch/home/2024/03/27/libhelix-behind-the-scene/?preview_id=6192&preview_nonce=8710762d36&_thumbnail_id=3546&preview=true). I was just testing with 80 bytes and it was working fine

VellingaT, 2024-03-27:
That is a quick response!
Unfortunately I can't access the content of the link you provided.
"Sorry, you are not allowed to preview drafts".

Few remarks:
- it was on a ESP32C3.
- it didn't happen at all the mp3 files. I used "Free_Test_Data_500KB_MP3.mp3" and a buffersize of 512 bytes per read/write action. Then the dropping of frames was quite large; with buffersize 4096 just once, with buffersize of 4000 no issues.

And another small change in "MP3DecoderHelix" around line 107: 

        // TV 26/03/2024: Small change, bytes to search reduced with offset to prevent checking outside the buffer
        int findSynchWord(int offset=0) override {
            int result = MP3FindSyncWord(frame_buffer+offset, buffer_size-offset); 
                                                                                                                 ======

pschatzmann, 2024-03-27:
https://www.pschatzmann.ch/home/2024/03/27/libhelix-behind-the-scene/

VellingaT, 2024-03-27:
Copied the files and ran the test after changing line 96 in "Buffers.h":
    LOG_HELIX(LogLevelHelix::Debug,"%s : %d", "LOG_METHOD", len);
LOG_METHOD was obviously not defined.

It now worked without any issues and I saw you already had changed the last remark also.

Thank you for the good work !

pschatzmann, 2024-03-27:
Thanks for your feed back: I was always testing this as part of the AudioTools, so I did not notice that LOG_METHOD is not defined if you compile it stand alone.

I committed the correction...

pschatzmann, 2024-03-30:
The new API has been merged

closed 2024-03-30 by pschatzmann

VellingaT, 2024-03-31:
LS,

using the lib standalone gave no compiler errors anymore. 
However, using different buffer sizes feeding the decoder produces a different number of output bytes so it looks something is going wrong. (I used "Free_Test_Data_500KB_MP3.mp3" and a buffersize of 512 bytes per read/write action => 795 callbacks 1831680 bytes; with buffersize 4000 => 801 callbacks 1845504 bytes; with buffersize 4096 => 798 callbacks 1838592 bytes).

pschatzmann, 2024-03-31:
I rather think this is a feature of the decoder. 

Independently of what buffer size you feed, the decoder will always get the same package size (MP3_MIN_FRAME_SIZE=1024 bytes) to decode.

I actually never use the decoder via the callback, but I just forward the data to the output device which is usually I2S

VellingaT, 2024-03-31:
Don't feel it is the decoder; as far as I can see it always expects/gets the same input which should result in the same output.
Outputting different number of bytes means different sound. 
As far as I can see you use the same moment to forward the data or to do a callback.
I will try to look in the source myself and come back to you if I find something.

VellingaT, 2024-03-31:
Figured it out; my mistake. 
I can confirm it now produces always the same output and no frame drops.
Case closed :)