pschatzmann/codec-helix #13: Version > 0.8.1 doew not work on plateformio

philippejadin, 2024-04-12:
First thank you for this great library. I hope my diagnostic is correct, it seems that latest version don't work on plateformio anymore.

Here is a minimal plateformio.ini file to reproduce : 
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
lib_deps = 
  https://github.com/pschatzmann/arduino-libhelix
  https://github.com/pschatzmann/arduino-audio-tools
[platformio]
src_dir = src
```

Errors I get : 
- .pio/libdeps/pico/libhelix/src/utils/helix_log.h:30:72: error: 'snprintf' was not declared in this scope
- .pio/libdeps/pico/libhelix/src/utils/helix_log.h:17:29: error: 'Serial' was not declared in this scope

It works fine if I pin the library versions to : 

```
lib_deps = 
  https://github.com/pschatzmann/arduino-libhelix#v0.8.1
  https://github.com/pschatzmann/arduino-audio-tools#v0.9.5
```

pschatzmann, 2024-04-12:
I committed a correction to libhelix, but I noticed that Platformio is messing up with the wifi information in the AudioTools:
Now you can also specify the compile option NO_WIFI to switch the WiFI off, otherwise you will need to provide the additional libraries for WiFi support as well

philippejadin, 2024-04-14:
I confirm it works fine now. Thank you for your time! I'm reconsidering platformio as a more reproducible environment for dev :)

closed 2024-04-14 by philippejadin