pschatzmann/codec-helix #14: Commit 6fcc29e seems to break setNotifyAudioChange callbacks

mikecomstock, 2024-04-15:
I'm using git@github.com:pschatzmann/arduino-audio-tools.git#v0.9.7 (though I don't think this version matters for this issue) on an esp32. When I use that in conjunction with https://github.com/pschatzmann/arduino-libhelix/commit/6fcc29eaeab7b75451ef55fe3183fc624a3e905f, none of the setNotifyAudioChange events seems to fire, so AudioInfo's don't get sent from the mp3 decoder, to volume, to the I2S speaker.

I don't quite understand the code in 6fcc29eaeab7b75451ef55fe3183fc624a3e905f so I'm not sure if I'm using the library incorrectly or if it's a bug.

Note that I'm using it with MP3DecoderHelix.

mikecomstock, 2024-04-15:
It may be that in MP3DecoderHelix.h, 

```
  void setInfoCallback(MP3InfoCallback cb, void *caller = nullptr) {
    this->infoCallback = cb;
    if (p_caller_ref!=nullptr)
      p_caller_ref = caller;
  }
```

should instead check for
`    if (p_caller_ref!=nullptr)`

pschatzmann, 2024-04-15:
Yes that might maybe make more sense.
However, I did not notice any issues in the actual audio tools version because the caller is set via a separate method call...

The correction has been committed...

mikecomstock, 2024-04-16:
That fixed it - thank you!

closed 2024-04-16 by mikecomstock