pschatzmann/codec-helix #21: aacPlus+ compatability?

mmackh, 2024-09-15:
Hello Phil,

is this version of the AAC decoder compatible with aacPlus? It seems to work somewhat with the lower bitrate, but crashes on the higher ones. MP3 version of the stream works with a few crackles (probably my system load)

http://stream.srg-ssr.ch/m/rsc_de/aacp_48 -> Works, but need to increase length *2? 
http://stream.srg-ssr.ch/m/rsc_de/aacp_96 -> Crashes

http://stream.srg-ssr.ch/m/rsc_de/mp3_128 -> Works with slight noise

pschatzmann, 2024-09-15:
Did you read the chapter about Parameters ?

mmackh, 2024-09-15:
I compiled Audio Tools and put the stream link in there - it worked. We have our own custom frontend, the issue must be there. Thanks for your response, have a good day!

closed 2024-09-15 by mmackh