pschatzmann/codec-helix #24: Calling AACDecoderHelix.end() intermittently gets stuck and never finishes

kel30a, 2025-02-17:
Calling AACDecoderHelix.end() intermittently gets stuck and never finishes.

Its getting stuck in the AACDecoderHelix.flush() [here](https://github.com/pschatzmann/arduino-libhelix/blob/41b6b0c31e5f773f5d66f8a2147a7087bb595209/src/AACDecoderHelix.h#L54).

I did some poking around and noticed that a call to the flush function is commented out in the arduino-audio-tools library [here](https://github.com/pschatzmann/arduino-audio-tools/blob/d78cd1c9c99d36b8ab7b01f85981ad4c7d6ac26a/src/AudioTools/AudioCodecs/CodecAACHelix.h#L139).

Is there a reason why it was commented out? Can it be safely be commented out in the AACDecoderHelix.end() too ?

Digging further in it appears the "bitsAvail" is getting set to a negative number [here](https://github.com/pschatzmann/arduino-libhelix/blob/41b6b0c31e5f773f5d66f8a2147a7087bb595209/src/libhelix-aac/aacdec.c#L376C48-L376C57) and [here](https://github.com/pschatzmann/arduino-libhelix/blob/41b6b0c31e5f773f5d66f8a2147a7087bb595209/src/libhelix-aac/aacdec.c#L387). Catching when this happens and just returning ERR_AAC_UNKNOWN seems to also avoid the issue.

pschatzmann, 2025-02-17:
I committed a correction which makes sure that in flush only complete segements are decoded

closed 2025-02-17 by pschatzmann

kel30a, 2025-02-17:
Its still happening for me, see attached stacktrace. I'll see if I can create a small reproducible snippet. I think it may happen after I have used a URLStream that timed out (due to network access)

[image]

pschatzmann, 2025-02-18:
I used the following test. If you think that there is still an issue, please provide a test case that let's me reproduce your issue!

```C++

#include <SPI.h>
#include <SD.h>
#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecAACHelix.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"

// files from https://filesamples.com/formats/aac
const char* filesNames[] = {
  "/aac/sample1.aac",
  "/aac/sample2.aac",
  "/aac/sample3.aac",
  "/aac/sample4.aac",
  nullptr,
};
int fileIdx = 0; 

const int chipSelect = PIN_AUDIO_KIT_SD_CARD_CS;
AudioBoardStream i2s(AudioKitEs8388V1);                                         // final output of decoded stream
EncodedAudioStream decoder(&i2s, new AACDecoderHelix());  // Decoding stream
File audioFile;
StreamCopy copier(decoder, audioFile);
uint64_t timeout = 0;;

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);
  LOGLEVEL_HELIX = LogLevelHelix::Debug;

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);
  config.sd_active = true;
  i2s.begin(config);

  // setup file
  SD.begin(chipSelect);
  audioFile = SD.open(filesNames[fileIdx++]);

  if (!audioFile){
    Serial.println("file does not exist");
    stop();
  }

  // setup I2S based on sampling rate provided by decoder
  decoder.begin();
}

void nextFile() {
  decoder.end();
  Serial.println("==> end");
  decoder.begin();
  // move to beginning on end
  if (filesNames[fileIdx]==nullptr) fileIdx = 0;
  audioFile = SD.open(filesNames[fileIdx++]);

  timeout = millis() + random(2000, 4000);
}

void loop() {
  if (millis() > timeout){
    nextFile();
  }

  if (!copier.copy()) {
    nextFile();
  }
}

```