pschatzmann/codec-helix #34: Playing mp3 with a lot of meta data causes mp3DecInfo->nSlots to be -1

liusida, 2025-09-17:
### Problem Description

Thank you for your libraries.

I can play simple mp3 audio files, but when i play some music mp3 files, the program crashes.

I digged into the code and found that this line causes nSlots to be -1 when handling this kind of mp3 files, and later when used in memcpy, it causes crash:

https://github.com/pschatzmann/arduino-libhelix/blob/5088412c02025a0b118b8bf4f01a9a922ae2a9b9/src/libhelix-mp3/mp3dec.c#L336

My IDE suggests me to remove all the metadata before playing, so it proposes this function for me to preprocess the mp3 file before playing, and it avoids the crash:

```
  auto skipID3v2 = [&](Stream& s){
    uint8_t hdr[10];
    if (s.readBytes(hdr, 10) != 10) return;                 // not enough -> ignore
    if (hdr[0]=='I' && hdr[1]=='D' && hdr[2]=='3') {
      uint32_t sz = ((hdr[6]&0x7F)<<21)|((hdr[7]&0x7F)<<14)|((hdr[8]&0x7F)<<7)|(hdr[9]&0x7F);
      uint32_t toSkip = sz;                                  // header already consumed
      uint8_t buf[256];
      while (toSkip) {
        size_t n = s.readBytes(buf, toSkip > sizeof(buf) ? sizeof(buf) : toSkip);
        if (!n) break;
        toSkip -= n;
      }
    } else {
      // Not ID3v2: we consumed 10 bytes; simplest workaround is to re-open the URL
      // so we don't lose them:
      url.end();
      url.begin(MP3_URL, "audio/mpeg");
    }
  };
  skipID3v2(url);
```
I'm not sure about MP3's file format... seems it is related to ID3.

I think you might want to prevent the crash caused by complex metadata, thus this report. Have a nice day.

### Device Description

ESP32 S3 Box 3
ES83111

### Sketch

```cpp
// not that important, but I pasted anyway.

#include <Arduino.h>
#include <WiFi.h>
#include <esp_wifi.h>
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"
#include "AudioTools/CoreAudio/AudioHttp/URLStream.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

static const char* SSID     = "";
static const char* PASSWORD = "";
static const char* MP3_URL  = "http://192.168.1.164:1234/output.mp3";

// Audio board and output (ES8311)
AudioBoard board{AudioDriverES8311, NoPins};
AudioBoardStream out(board);
AudioInfo info(32000, 2, 16);

// Network stream and MP3 decoder pipeline
URLStream url;                    // HTTP client stream (uses Arduino WiFi)
MP3DecoderHelix mp3;              // MP3 decoder
EncodedAudioStream decoded(&out, &mp3); // writes decoded PCM to I2S output
StreamCopy copier(decoded, url, 4096);

void audioTask(void*){
  for(;;){
    copier.copy();
    vTaskDelay(1);
  }
}

void setup() {
  Serial.begin(115200);
  while (!Serial) { delay(10); }

  WiFi.mode(WIFI_STA);
  WiFi.persistent(false);
  WiFi.setSleep(false);
  WiFi.setAutoReconnect(true);
  WiFi.setTxPower(WIFI_POWER_19_5dBm);   // boost TX power

  Serial.printf("Connecting to %s\n", SSID);
  WiFi.begin(SSID, PASSWORD);

  uint32_t t0 = millis();
  while (WiFi.status() != WL_CONNECTED && millis() - t0 < 20000) {
    delay(250);
    Serial.print(".");
  }
  Serial.println();
  if (WiFi.status() != WL_CONNECTED) {
    Serial.println("WiFi connect timeout");
    return;
  }

  Serial.print("WiFi connected. IP: ");
  Serial.println(WiFi.localIP());
  
  // Configure audio pins for ESP32-S3 Box 3
  Serial.println("initializing audio pins...");
  DriverPins pins;
  pins.addI2C(PinFunction::CODEC, 18, 8);           // SCL=18, SDA=8
  pins.addI2S(PinFunction::CODEC, 2, 17, 45, 15);   // MCLK=2, BCLK=17, WS=45, DOUT=15
  pins.addPin(PinFunction::PA, PA_PIN, PinLogic::Output);
  board.setPins(pins);

  // Start audio output
  Serial.println("starting I2S & codec...");
  out.begin();
  out.setAudioInfo(info);
  out.setVolume(0.7f);

  // Start network stream and decoder pipeline
  Serial.println("opening URL stream...");
  url.setTimeout(10000);
  if (!url.begin(MP3_URL, "audio/mpeg")) {
    Serial.println("URL open failed");
    return;
  }
  // Initialize encoded stream so decoder can push PCM to output
  decoded.begin();
  copier.begin();
  // Avoid overrunning target by honoring availableForWrite()
  copier.setCheckAvailableForWrite(true);
  xTaskCreatePinnedToCore(audioTask, "audio", 4096, nullptr, 10, nullptr, 0);
}

void loop() {
  // no-op
}
```

### Other Steps to Reproduce

I guess there might be license issue with uploading music mp3 files to the public. If you can't reproduce, I can send that mp3 file through email.

[image]

### What is your development environment (with Arduino Core Version information)

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32s3box3.html
PLATFORM: Espressif 32 (55.3.30) > Espressif ESP32-S3-Box-3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - contrib-piohome @ 3.4.4 
 - framework-arduinoespressif32 @ 3.3.0 
 - framework-arduinoespressif32-libs @ 5.5.0+sha.b66b5448e0 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.5.43
|-- lvgl @ 9.1.0
|-- GT911 @ 0.2.1+sha.85875b2
|-- NimBLE-Arduino @ 2.3.6+sha.dfb4ac5
|-- audio-driver @ 0.1.4+sha.7026dc2
|-- audio-tools @ 1.1.3+sha.e2d02aa
|-- libhelix @ 0.8.9+sha.17c28c8
|-- WiFi @ 3.3.0
|-- Preferences @ 3.3.0
|-- Wire @ 3.3.0
Building in release mode

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-09-17:
Did you try to [filter out the metadata as documented here](https://github.com/pschatzmann/arduino-audio-tools/wiki/Audio-Metadata#metadata-and-decoders) ?

liusida, 2025-09-18:
I should. Thanks!